MichielVanwelsenaere/HomeAutomation.CoDeSys3 #33: MQTT connected not correct

YorickErnst, 2019-12-03:
Hi Michiel,

I've started working with your latest release, and would like to make a split scenario where one instruction is executed if no MQTT connection is made and another instruction is done if there is an MQTT connection.

At the moment all messages end up fine in my MQTT broker but the MQTT_CONNECTED value in MQTTInfo remains FALSE.

MichielVanwelsenaere, 2019-12-04:
Hi,

The MQTT_CONNECTED in MQTTInfo becomes high for one PLC cycle when there's an initial connector and when there's a reconnect to the MQTT broker.

Nevertheless, it does make sense to have a boolean indicating the connection status to the broker for purposes like you are indicating. I'll check what I can find but might need to redirect that question to Stefan (MQTT lib developer).

MichielVanwelsenaere, 2019-12-04:
Just gave this a look, the boolean below under the MQTTClient object seems what you need:
[image]

let me know if that answers your question.

YorickErnst, 2019-12-04:
Perfect, that's indeed what I need :)

closed 2019-12-04 by YorickErnst