MichielVanwelsenaere/HomeAutomation.CoDeSys3 #43: Comparison with SNDLTN libraries

Sandolution, 2019-12-17:
Comparison with already running project using the following libs:
[image]

These work together like this:
[image]

Attached is a .projectarchive (inside ZIP), should be able to import this in E!Cockpit or CoDeSys.
Open libs are in it.
[http://sandolution.com/wp-content/uploads/2019/12/Pallas.zip](url)

MichielVanwelsenaere, 2019-12-17:
great, thanks! Going to try to find some time for this!

MichielVanwelsenaere, 2020-01-27:
Managed to have a brief look at this, not nearly as deep as I'd like to but enough to start a conversation on merging codebase.

How do you look at this high level? (Let's try to find common ground :) )
 - The high level approach of this project is to easily set and get values from the PLC but do the complex integrations and automations in a higher level software like Home Assistant, Domoticx, etc.. Do you have the same vision
- Merging codebase means opensourcing your libraries, are you up for that?
- I would avoid doing a cyclic http update as it might trigger some automations unwanted in some home automation software packets. If it is about maintaining state with 100% certainty, why not use the MQTT retain flag?
- The architecture between the projects is fundamentally different, how do we get to one architecture and what is it? 
- Priority: with construction coming closer HVAC support and a modbus environment sensor has prio on my (personal) end. Thoughts?

MichielVanwelsenaere, 2020-03-24:
Going to close this one due to inactivity. Feel free to reopen.

closed 2020-03-24 by MichielVanwelsenaere

Sandolution, 2020-11-02:
Sorry for my absence since I posted this, haven't had the time to pick this up any further.
However, with the 'darker days' approaching, I have some more spare time on hand.

To start with, I upload my libraries to GitHub:
https://github.com/Sandolution/CoDeSys_Domoticz

Since I think we're definitely not 'competitors' and can exist perfectly next to each other, as we already did before, feel free to look at my approach so far. And I will definitely have a closer look at the things you already did.