MichielVanwelsenaere/HomeAutomation.CoDeSys3 #52: codesys x64 bits libraries sysfile23 and cmperrors interfaces

runtimevic, 2020-01-08:
hello, 

I have a problem with 2 libraries that tells me they are only for 32 bits:
sysfile23 and cmperrors interfaces

how can it be solved?
best regards, Víctor.

MichielVanwelsenaere, 2020-01-08:
Hi Victor,

As discussed this seems related to the 64bit architecture of your beckhoff PLC, with the OSCAT network library only supporting 32bit.

Perhaps Stefan has an idea on this one, adding the related issue on his repo here for traceability:
https://github.com/stefandreyer/CODESYS-MQTT/issues/32

MichielVanwelsenaere, 2020-01-27:
Seeing the updates on the issue on Stefan's repository I do suppose this one can be closed @runtimevic ?

closed 2020-02-08 by MichielVanwelsenaere

MichielVanwelsenaere, 2022-06-03:
Hmmm, I must admit that I've only been running this on a 32 bit device. I'll need to do some digging here.

For clarity, is this the only lib giving trouble? What device are you using?

nehaltrivedi97, 2022-06-05:
So actually I am working on an old program of a previous student. This is for my thesis. He had written the code in Codesys V3.5 SP11 and i guess it was a 32 bit device. I am working with a 64 bit device. I am using a IFM AC1421 PLC. I mean there is no code in the error. So this error occurs once i enter the simulation mode. Otherwise it runs fine.

erdemustunel, 2023-03-14:
Hello,
is the problem solved , because I have faced this problem , and I can not progressing.

Thank you.

Erdem Üstünel