MichielVanwelsenaere/HomeAutomation.CoDeSys3 #54: Subscription Issue for Custom Project

Araforn, 2020-03-16:
Hello,

I'm raising this issue based on my initial use of Stefan Dreyers library. See my posts here.

https://github.com/stefandreyer/CODESYS-MQTT/issues/33

I really want ton  use the MQTT architecture you have set up for your project but have issues creating a new subscription.

**My Use Case (Copied from the issue posted above)**

Basically I have a Codesys project that I developed to interface with Samsung Smart things.

https://community.smartthings.com/t/raspberry-pi-codesys-project/79308/32

Separately I developed a python based web socket smart app that would write events to the codesys OPC UA server.

https://community.smartthings.com/t/smart-app-multiple-device-subscriptions/159166
This all worked but had limitations.

I since moved to Home Assistant and I discovered MQTT. In home assistant I have "Statestream" setup where it will publish all events from my instance of Home Assistant to MQTT topics.

https://www.home-assistant.io/integrations/mqtt_statestream/

I need my codesys project to subscribe to these topics which I will then use these strings to populate all the devices in my codesys project. This all works already using my smart things integration but now i have moved to Home assistant and MQTT. Essentially I just need to read all these event/state strings over MQTT using a couple of specific subscription topics.

**My issue**
I have imported your project and all libraries. I want to simply add a new subscription so I have followed your HowTo 

https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3/blob/master/docs/FAQ/Howto_adding_a_new_MQTT_subscription.md

I have did everything except step 5..... 

**"Register the required function blocks against the callback function block to trigger the callback method when a new message arrives on the subscribed topic. Example below:
pMqttCallbackCollector^.put(instance:= THIS^);"**

Where do I do this?. 

 
Here are some screens of my new Subscription FB based on your HowTo for subscriptions
 Step 1 ( I have created a new lwt and birth topic in this example)
[image]
)

Step 2
[image]

Step 3
[image]

Step 4 (I create a custom Topic "Barden_PLC")
[image]

Step 5 
????

Step 6 (Copied one of your FB's and renamed it)
[image]

[image]

If i have done everything right I should see any new payload on this topic "Barden_PLC" is this correct??. Getting this working will allow me to continue with my integration on my own project.

 Note that when on online I do connect to my broker and get the "online" status as per below.

[image]

Could you let me know how to implement step 5 and based on the images above perhaps you are seeing something else I am doing wrong? Your help would be much appreciated here. Thanks so much in advance.

Best Regards,

Paul.

MichielVanwelsenaere, 2020-03-16:
Hi Paul,

That's quite the project :wink:

I don't have my laptop with me right now but what you're looking for is inside the InitMqtt method of any MQTT function block in the project.

Let me know if that answers your question, I'll review that piece of documentation and see if it can be improved.

Best regards,
Michiel

stefandreyer, 2020-03-17:
Hi Paul,

you need to add your instance of FB_BARDEN_MQTT to the collector too:
collector_BARDE.put(instance:=(instance of FB_BARDEN_MQTT));

Then you PublishReceived method should be called when a packet arrives...
Test this with a brakepoint...

BR Stefan

Araforn, 2020-03-17:
Thanks for your replies guys but perhaps my more in depth knowledge of codesys and methods is leaving me down because it still doesn't work for me.

The instance name of my FB is FB_BARDEN_TEST as below
[image]

1. I've stripped down my FB to just the following. No logic in here currently because i'm only interested in the data received by the subscription.
[image]

2. InitMqtt - Note I can't add explicitly.

collector_BARDEN.put(instance:=FB_BARDEN_TEST);

This is because my imported FB's are within the POU section in codesys which does not allow me to add my global variables. However using.

pMqttCallbackCollector^.put(instance:= THIS^) should work here anyway unless I am missing something?

[image]

3. Publish Received
[image]

When I go online I can receive the "online" birth message as usual but I do not receive any data on my FB when I issue an external client to publish to my custom topic. I define the topic  here. "Barden_PLC"
[image]

I insert the breakpoint in the below position in the publish received method but it is never hit when that topic is published to.
[image]

Bearing in mind my only objective is to receive the string data on this topic from this FB and start populating other strings when I eventually bring this functionality into my project.

Can you see what I am missing?.

Thanks again for your help with this.

Regards,

Paul.

stefandreyer, 2020-03-18:
Hi Paul,

If you want to use only one FB for Receiving data you don't have to use the collector. The collector is for use for more then one instance which have to do the receive work. If so, you can register your FB directly to the input of the subscription FB(callback:=PLC_PRG_MAIN.FB_BARDEN_TEST).

If you use pMqttCallbackCollector^.put(instance:= THIS^) 
then you have to register this collector on your subscription FB(callback:="MqttCallbackCollector").

collector_BARDEN.put(instance:=FB_BARDEN_TEST); 

Must work to, I'm doing it like this in my projects. Try collector_BARDEN.put(instance:=PLC_PRG_MAIN.FB_BARDEN_TEST); But call put only once, or it will be added to the list in each cycl until dynamic memory is out.

You may show me the data you want to receive? (Topic, Data)

BR Stefan

Araforn, 2020-03-18:
Hi Stefan,

Thanks for the help again.

For the purpose of my understanding I just want to look at one topic and one FB. I just want to get one FB working correctly first so i know its working in this example project. 

However for my main project I will look at multiple topics so we will need multiple FB's and therefore will need to use the collector.

Please see below in response to inserting collector_BARDEN.put(instance:=PLC_PRG_MAIN.FB_BARDEN_TEST). It returns an error when using this global collector.

[image]

Therefore i will use  pMqttCallbackCollector^.put(instance:= THIS^)
 in which case I already register this collector on my subscription FB(callback:="MqttCallbackCollector")." See below.

[image]

Perhaps the easiest way for me to show you is to export my project which is attached below.

[MQTT_DEVELOPMENT_Update_17MAR2020.zip](https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3/files/4348732/MQTT_DEVELOPMENT_Update_17MAR2020.zip)

For testing I am only using the topic "Barden_PLC" and using an external client to post a payload "ON/OFF etc.....". It doesn't really matter about the payload once I can start to receive it.

Apologies if this is more my lack of understanding fundamentally

stefandreyer, 2020-03-18:
Hi Paul,

you don't call InitMQTT of your FB_BARDEN_MQTT instance.
I think you need to add:

[image]

BR Stefan

Araforn, 2020-03-19:
Stefan,

This worked!. Thank you so much for your assistance. I can now receive the payload when i insert the Breakpoint.
[image]

I have also checked against my home assistant statestream and there is no crashes. events are streaming in. This is fantastic. Thanks for your work in this library.
 i just have 2 more question if you could finally clarify.

1. Since the events are streaming in quite fast, i will need to buffer them.  Are previous received  subscription/payloads stored in an array somewhere so i can look back on what was received?

2. My lights topic in home assistant uses a wild card '+' : **hass/light/+/state**
This returns the state of any light device. Using an external MQTT client i see the full topic when i subscribe. See below for example. Here i have subscribed to **hass/light/+/state** but it returns the state (off)and the actual topic **hass/light/actual light/state**. is this possible in your library?

[image]

Thanks once gain for your help. i can close this topic regardless.

Araforn, 2020-03-19:
Thanks also to Michiel for providing this great architecture and documentation that allowed me to make this work.

Really great work and thanks for sharing.

closed 2020-03-19 by Araforn

stefandreyer, 2020-03-19:
Hi Paul,

1. there is no Buffer back, if the Client receives a publish packet the Content is directly delivered to the callback attaced to the callback instance of the subscription FB. Due this the Client is able to handle a high freqency of packets because it doen't handle just one packet per cycle. The Client is able to handle 50 QoS 1 or 2 packets at the same time. This should be a Parameter in your broker too. 

2. You can make a subscription to **hass/light/+/state** and will receive **hass/light/porch/state**, in the PublishReceived method you will allways get the non wildcard topic. This behaviour is exactly like your MqttFX client. Just check out the data in the data struct when you set the breakpoint.

BR

Stefan