MichielVanwelsenaere/HomeAutomation.CoDeSys3 #67: Add pulse output

domoticafan, 2020-04-30:
First of all, thanks for this great project. I was able to get it work on a 750-8212 without any PLC/e-cockpit knowledge.

Beside the output high/low toggle I would like to control my bi-stable relay which need a pulse signal. So can you add also a 'pulse' output feature?
I did some testing with the TOF timer, however I have no idea if this is the best way to do it.

TOFInst(IN := PULSE, PT:= T#1S);
OUT := TOFInst.Q;

MichielVanwelsenaere, 2020-04-30:
Hi,

Good to hear that you were able to get this working swiftly.

Problem with a bistable relais is that it's not possible to know the state of the relais in the PLC. 

So how would you know in your home automation software what the state is of your device?

Technically I can make a pulse output FB, I just don't see how it would work functionally because you don't know the eventual state of the relais? 

Let me know your thoughts!

domoticafan, 2020-04-30:
I will use an interface relay or aux contact to detect the status of the relay. Yes it is quite 'expensive' to use an extra input for the status. However, in this case I'm able to control the most important lights without any automation (hass/plc/whatever).

MichielVanwelsenaere, 2020-05-01:
I'll look into this. Idea would be to create a new FB 'FB_OUTPUT_SWITCH_MQTT' which has configurable pulse duration at the output and can be triggered by:

1. a pulse on the input
1. a 'Pulse' MQTT message on the FB 'in' MQTT topic

If the FB generates a pulse on its output it will generate a 'pulse' messsage on the FB 'out' topic.

Sounds good?

primsam, 2020-05-01:
Hello,

I think that this would be a very special Solution and not so popular even?
Maybe a simple way to do it can be to use the some output from the FB_INPUT_PUSHBUTTON_MQTT and on the feedback the FB_INPUT_BINARYSENSOR_MQTT to check the state of the relay.

domoticafan, 2020-05-01:
> I'll look into this. Idea would be to create a new FB 'FB_OUTPUT_SWITCH_MQTT' which has configurable pulse duration at the output and can be triggered by:
> 
>     1. a pulse on the input
> 
>     2. a 'Pulse' MQTT message on the FB 'in' MQTT topic
> 
> 
> If the FB generates a pulse on its output it will generate a 'pulse' messsage on the FB 'out' topic.
> 
> Sounds good?

Sounds good to me.

primsam, 2020-05-01:
Hello again

I made some thoughts again mabe it could be useful for someone if there exists a FB_OUTPUT_PB_MQTT?
I just tested it and for me it can make sense for some cases .
please can someone take a look at it ?

[FB_OUTPUT_PB_MQTT.zip](https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3/files/4564812/FB_OUTPUT_PB_MQTT.zip)

MichielVanwelsenaere, 2020-05-02:
Yes, I'll have a look!
But if I'm getting things right idea would be to incorporate the bistable relay feedback to the FB so the state is known an you can still send 'ON'/'OFF' MQTT messages to the FB to control the bistable relay?

MichielVanwelsenaere, 2020-05-02:
still need to check your codesample but could you both check if this docs page describes the behaviour FB that you require:
https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3/blob/feature/FB_OUTPUT_BISTABLE_MQTT/docs/FunctionBlocks/FB_OUTPUT_BISTABLE_MQTT.md

MichielVanwelsenaere, 2020-05-02:
@primsam , looking at your FB I see no feedback input. You guys do plan to wire a feedback loop back into the PLC right? If so, I would have a FEEDBACK input in the FB and control the state from in the FB.

domoticafan, 2020-05-02:
> But if I'm getting things right idea would be to incorporate the bistable relay feedback to the FB so the state is known an you can still send 'ON'/'OFF' MQTT messages to the FB to control the bistable relay?

That's correct.

MichielVanwelsenaere, 2020-05-08:
I made a branch that has a FB_OUTPUT_BISTABLE in the codebase. My initial tests indicate that it is working. Can you guys do a test as well?
Branch: [link](https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3/tree/feature/FB_OUTPUT_BISTABLE_MQTT)
FB docs: [link](https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3/blob/feature/FB_OUTPUT_BISTABLE_MQTT/docs/FunctionBlocks/FB_OUTPUT_BISTABLE_MQTT.md)

domoticafan, 2020-05-14:
Just did some tests and it seems to work correctly. Thanks for the very fast response and fix.

closed 2020-05-14 by domoticafan

primsam, 2020-05-17:
Unfortunately i can not test it becouse i don´t have any hardware for it .

PR #69 (2020-05-15): adding FB_OUTPUT_BISTABLE_MQTT
## Pull Request Checklist
Check if the tasks below are relevant for your PR, to mark an item as completed use *[X]*.

* [X] Updated the PLCOpen XML export file (only relevant on code change).
* [X] Updated the CoDeSys V3 export file (only relevant on code change).
* [X] Updated or created any relevant documentation.

## Proposed Changes

Adds a FB_OUTPUT_BISTABLE_MQTT function block to control bistable relais with on/off commands. Accomplished by a wired feedback loop back into the PLC so the software is aware of the state

## Related issues

- Closes #67