MichielVanwelsenaere/HomeAutomation.CoDeSys3 #70: Feature request: basic function block: FB_INPUT_MOTIONSENSOR_MQTT

ArjanM83, 2020-05-15:
When testing the Panasonic AMN31112J motion sensor I noticed that the output changes to HIGH when motion is detected. But quickly again it goes back to LOW, and it fluctuates quickly in situations of change. When a person walks into a room this could result in 10-100 times HIGH within a couple of seconds.

To avoid excessive traffic between the PLC and the MQTT broker, it would be nice to have a basic function block designed specific for motion sensors. The following rules could be an example of its behaviour:
- When detecting motion the state becomes HIGH.
- When no motion has been detected for at least 5 seconds the state becomes LOW.
The number of seconds could be provided as a parameter of the function block.

@MichielVanwelsenaere has referred to the concept of hysteresis.

MichielVanwelsenaere, 2020-05-17:
Thanks for the feedback with your findings! Would it also make sense to you to implement an additional parameter to the binary sensor FB that allows for configurable hysteresis on the output?

What are you by the way using to transform the 5V signal from the motion sensor to something the PLC can read?

ArjanM83, 2020-05-17:
> Would it also make sense to you to implement an additional parameter to the binary sensor FB that allows for configurable hysteresis on the output?

Can you provide an example of how this would work?
 
> What are you by the way using to transform the 5V signal from the motion sensor to something the PLC can read?

I'm using the Wago 753-434 with 8 digital inputs, receiving power from the 750-612, wired to an external power supply (5v DC).

MichielVanwelsenaere, 2020-05-18:
I would create a method 'ConfigureFunctionBlock' that can be called like below:

```
FB_MOTION_001.ConfigureFunctionBlock(hysteresis:=T#5S)
```

by default it would be disabled, by calling the method it would be configured (only required once during startup).
Behavior in this case would be that the output remains high for 5 seconds after the last input pulse.
For example: if you have 20 pulses over a period of 5 seconds, the output would be high for 10 seconds.

Let me know your thoughts!

ArjanM83, 2020-05-18:
> Behavior in this case would be that the output remains high for 5 seconds after the last input pulse.
> For example: if you have 20 pulses over a period of 5 seconds, the output would be high for 10 seconds.

Implementation and behavior sounds good!

DomiGijzen, 2020-05-19:
I was picking up my PLC project again and started with the same sensor and noticed the same behavior. 

 

This solution sounds good!

MichielVanwelsenaere, 2020-05-22:
I enhanced the binary sensor input FB so it can be configured with a 'turnoffdelay'.
Code can be found here; [link](https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3/tree/feature/binarysensor-hysteresis)
Updates docs can be found here; [link](https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3/blob/feature/binarysensor-hysteresis/docs/FunctionBlocks/FB_INPUT_BINARYSENSOR_MQTT.md)

But in short, the below is what you need to add: 
`FB_INPUT_BINARYSENSOR_MQTT.ConfigureFunctionBlock(T_TurnOffDelay:= T#5S);`

Could you let me know when you've found some time to test this and the implementation is behaving as you would expect so I can merge this to master branch?

ArjanM83, 2020-05-22:
Thanks for your quick update!

The code and docs look good. After my recent tests I have 'dismantled' my PLC test situation. My plan is to continue testing again at the end of the year. My project is long term, the house to implement this project in will be finished in 2021 or 2022.

Maybe @DomiGijzen could test the new functionality? Just an idea, I don't know if he has time available..

primsam, 2020-05-23:
I have a similare use in my project and used a near identical FB like that .
So it was a drop in replacement and work as expercted for me.
Thank you very much for the work

MichielVanwelsenaere, 2020-05-24:
Perfect! Thanks for checking!

closed 2020-05-25 by MichielVanwelsenaere

DomiGijzen, 2020-06-13:
I saw someone checked. Git to it just today and it is also working in my code. Tnx!

PR #71 (2020-05-25): Adding a configurable turn off delay to the binary sensor input FB
## Pull Request Checklist
Check if the tasks below are relevant for your PR, to mark an item as completed use *[X]*.

* [X] Updated the PLCOpen XML export file (only relevant on code change).
* [X] Updated the CoDeSys V3 export file (only relevant on code change).
* [X] Updated or created any relevant documentation.

## Proposed Changes

- Adds a 'ConfigureFunctionBlock' method call to FB_INPUT_BINARYSENSOR_MQTT. This allows for a configurable delay upon which the output will become low after the input received a negative edge  [... cut]