MichielVanwelsenaere/HomeAutomation.CoDeSys3 #76: Codesys V3.5 SP16  Oscat library (basic)

thomkedc, 2020-11-13:
Hi,

The Basic library under 'BUILDING (OSCAT)  needs to be updated.
the updates that needs to be done are minor  but since i don't have the source from the building library (only the BASIC can i find on the OSCAT website)  I can add an new BASIC library, but then it's not included into the 'Building' Library'

See image below: (so the circled one is the library without any errors, this once should replace the basic library in the building folder )
[image]

MichielVanwelsenaere, 2020-11-13:
hmmm, that's a painful dependency problem..
what Codesys version are you using? this seems tied to a newer codesys version that introduces a new datatype 'LDT'.

My suggestion: try to get your hands on an older Codesys version to get you unblocked 

In the meantime we'll fire up a project to have our own library that contains the most important functionalities to get rid of any dependency problems (now and in the future).

drled, 2020-11-23:
I'm also using codesys V3.5 SP16 and cannot download the missing libraries needed for MQTT.
I'm not sure wether this is related to the same dependency problem?

[image]

MichielVanwelsenaere, 2020-11-23:
are all those library issues related to the MQTT library?

drled, 2020-11-23:
Thanks for your quick feedback.
I have errors in the following libs:
CommonTypsFunctions (Stefan Dreyer)
OSCAT Network(ie. SysFile23, SysSocket23)
SD_MQTT(CAA_File,CmpCrypto,...)

[image]
[image]

MichielVanwelsenaere, 2020-11-24:
CommonTypsFunctions (Stefan Dreyer) -> could you open an issue on his repo for these? feel free to link/mention it over here as well. Time to get these dependency problems sorted out 😄 
OSCAT Network(ie. SysFile23, SysSocket23) -> the two you are describing are normally under the 'libraries' folder in the solution, could you import them?
SD_MQTT(CAA_File,CmpCrypto,...) -> thanks for the complete list, I'll check if I can find these myself and create a library repo with everything in it

For all clarity: there's work ongoing to limit the amount of library dependencies this project has but it's only just getting started.

MichielVanwelsenaere, 2020-11-26:
@drled , do you still have some dependency problems? 
Otherwise, you can leverage this repo to get your missing files:
https://github.com/OpenHomeToolkit/CoDeSys3.LibraryRepo

let me know if that helps.

drled, 2020-11-27:
I was able to resolve most of the issues.

Now what's left is the same issue that was mentioned by @thomkedc. The LDT output tag in the RTC_2 functionblock of the Oscat basic lib is being recognized as a datatype.
[image]

MichielVanwelsenaere, 2020-11-27:
That was to be expected unfortunattely.
Work has started to extract & refactor the required piece of the OSCAT libs yet it's going to take some time..

Same remark as for Thomke, can you download an older version of Codesys to get you unblocked?

drled, 2020-11-27:
I installed codesys V3.5 SP15 and got it up and running without any dependency issue. Thank you!

rmeekers, 2021-05-30:
any updates on this issue?

MichielVanwelsenaere, 2021-06-01:
> 
> 
> any updates on this issue?

actually, yes! 
I asked a bit around and normally Codesys SP16 should come with an adjusted version of the OSCAT libs avoiding the compilation problem above. Haven't found time to verify yet however.

Feel free to look for them of you've got SP16 or higher installed.

MichielVanwelsenaere, 2021-06-01:
you can find the updated libs here:
https://store.codesys.com/catalogsearch/result/?q=OSCAT

I'll incorporate them in the project as the default when I have a moment.

PR #99 (2022-01-31): Update to support CodeSys 3.5 SP16
## Pull Request Checklist
Check if the tasks below are relevant for your PR, to mark an item as completed use *[X]*.

* [X] Updated the PLCOpen XML export file (only relevant on code change).
* [X] Updated the CoDeSys V3 export file (only relevant on code change).
* [ ] Updated or created any relevant documentation.

## Proposed Changes

- Update OSCAT libraries to latest versions (supporting CodeSys 3.5 SP16):
   - OSCAT Basic 3.3.4
   - OSCAT Building 1.0.0.1
   - OSCAT Network 1.3.5.3
- Update CODESYS-MQTT and CODESYS-Common to the latest versions:
   - CO [... cut]
closed 2022-01-31 by MichielVanwelsenaere