MichielVanwelsenaere/HomeAutomation.CoDeSys3 #78: Possibility to bundle the function blocks as a library

roomboot, 2020-11-23:
Hi,

I was wondering if it is possible to work with versioned libraries instead of projects to make upgrades easier. Is there a good reason not to do this?

MichielVanwelsenaere, 2020-11-23:
Hi,

there actually is:
- the whole project is more of a reference architecture, it's hard to abstract all the function block logic in one library as that wouldn't reflect the architectural structure required to use them. 
- libraries have their problems/challenges as well

Perhaps the question needs to be posed differently? 

> what's the recommended way to upgrade the project's function blocks swiftly? 

let me know your thoughts.

roomboot, 2020-11-23:
Yes, making the process of upgrading the function blocks easier and quicker is the goal of this issue. My first idea to achieve this is to use a library to bundle the POU's, to import and instantiate them and use as before. In that case upgrading should be as easy as just replacing the library dependencie(s) and optionally the breaking changes in the api. Can you clarify how the current architecture would limit this?

MichielVanwelsenaere, 2020-11-23:
the architecture doesn't limit this. What I'm trying to say by involving the architecture is the following:
1. a library is normally a collections of components (function blocks, functions, etc..) that can be imported in different projects for usage. They can in words to be considered as  'standalone components'
2. when abstracting all the POU's to a library from this project they can't function as standalone components as they require the queue, MQTT workers etc to work... (this is the architectural related piece) so it doesn't really make sense.

But as this is more related to ease of upgrading a library does not have to be used.
How about the following procedure to upgrade all POUs:
1. remove all existing POUs from the project
2. reimport all the POUs by using the PLCOpen xml

if possible, please give it a try and let m know how it goes!

roomboot, 2020-11-25:
Got it, I'll try to see how that works.

MichielVanwelsenaere, 2020-11-25:
let me know if it works so I can document a procedure for other users.

roomboot, 2020-11-30:
Seems to work. It's a bit clumsy, but better than manually copying.

I would suggest not deleting all current POU's but rather importing new items over the existing POU's. The user is prompted to replace the existing item(s). Doing it this way you can preserve the folder structure for folders like `Baselib`.

PR #80 (2020-12-15): Adding 'how-to' procedure to guide users to update function blocks to the latest version
## Pull Request Checklist
Check if the tasks below are relevant for your PR, to mark an item as completed use *[X]*.

* [ ] Updated the PLCOpen XML export file (only relevant on code change).
* [ ] Updated the CoDeSys V3 export file (only relevant on code change).
* [X] Updated or created any relevant documentation.

## Proposed Changes

- Adding 'how-to' procedure to guide users to update function blocks to the latest version

## Related issues

- Closes #78
closed 2020-12-15 by MichielVanwelsenaere