MichielVanwelsenaere/HomeAutomation.CoDeSys3 #81: Ability to process JSON MQTT string

mpvries, 2020-12-22:
Hi Michvw!

As you know I have your software running for a while without any problems and I'm very happy with it! Many thanks for all your contributions so far. 

I hava just a question, maybe it's simple to answer / implement. I use your library primariliy for lighting purposes. DMX is sent out from the PLC directly to an ARTnet module and all additional intelligence is built into Home Assistant / Node Red. MQTT is the protocol to echange al status information. 

Speaking of Home assistant: HA has the possibility for three types of MQTT Lights (see: [https://www.home-assistant.io/integrations/light.mqtt/#json-schema](url). 

I am currently using the 'default' scheme but I was wondering whether it is possible to add functionality for the JSON scheme into you code. With the JSON scheme it, for example, possible to use transitions and to flash lights. I've already tried to use the Wago JSON parser library but my programming knowledge is not enough to implement this sucessfully.

Do you think something like this is possible?

Many thanks for your answer!

MichielVanwelsenaere, 2020-12-24:
Hi!

Glad to hear about your positive experiences with my project! 

I personally don't have any experience with parsing json in codesys, if the library from wago doesn't help you any further perhaps you can give this one a try:
https://github.com/stefandreyer/JSON-Library

If you run into any issues or have questions, I'm sure Stefan will be able to help you out. You could also contact Wago support with your troubles with Wago library. Normally they are very helpful. 

Parsing the json is of course only the first step, you'll probably need to adjust your existing logic to a statemachine (if not already so) to play the received scenario's (transitions, light flashes, etc).

Not impossible but quite some work for sure.

If I were you I'd try receiving a json first, then build a working statemachine with hardcoded scenario and glue the pieces together once it's working.

Good luck!

FYI: I don't see any place for this in the main project as you're probably using a DMX library specific to Wago/é!COCKPIT and the project aims to be independent from a specific development environment (Codesys/é!COCKPIT).

MichielVanwelsenaere, 2021-01-25:
closing this due to inactivity. I do respond to closed issues.

closed 2021-01-25 by MichielVanwelsenaere