MichielVanwelsenaere/HomeAutomation.CoDeSys3 #84: Errors in project

Dorme, 2021-03-08:
I've been following the installation video's for a few times now, but I still keep getting errors in my project.
I'm using a Wago PFC200 with Codesys v3.5 SP16 Patch 3.

[image]

[image]

There are no errors in the Library Manager:

[image]

Also Codesys Control for PFC200 is correctly installed:

[image]

Anyone can point me towards the correct direction to solve this issue?

MichielVanwelsenaere, 2021-03-08:
Hi Dorme,

let's start with trying to get the basics to work. 
- Let's start with disabling the RS485 task (remove it from the taskconfiguration):
[image]

- could you show me the errors in 'PLC_PRG_MAIN'? that's the main PRG that is most straight forward so I'd like to start with getting that one to work.

In addition; some problems have been reported with this project and compatibility with Codesys v3.5 SP16 (#76). If this is costing you to much time effort, perhaps also consider a downgrade until I find some time to check the compatibility problems with SP16.

Dorme, 2021-03-08:
When switching to Codesys v3.5 SP15, I manged to get more stuff working. When leaving out the RS485 task, I can build and upload the project to the PLC. When I enable the RS485 task, I get theses errors:

[image]

MichielVanwelsenaere, 2021-03-08:
Great! For RS485 in Codesys 3S you need to do some setup. These docs should help you out:
https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3/blob/master/docs/RS485/UsingModbusRTU_CODESYS3S.md

Dorme, 2021-03-08:
If I'm not using the RS485 port on the PLC, I assume I can safely delete this task?

MichielVanwelsenaere, 2021-03-08:
indeed. It's only there as a 'boilerplate' for you. My FYI; it comes in mighty handy for reading out energy meters and all kinds of sensors in a reliable way. Naturally all setups are unique, just something to keep in mind.

Dorme, 2021-03-08:
Thanks for the clarification. I'll keep it in mind. Swithing to Codesys v3.5 SP15 did the trick.

closed 2021-03-08 by Dorme