MichielVanwelsenaere/HomeAutomation.CoDeSys3 #110: rs485_eastron_sdm220 RequestMaster method doesn't exist?

drled, 2022-08-03:
I'm trying to implement the FB "FB_RS485_EASTRON_SDM220" in Codesys but i get an error on the RequestMaster method:
Seems like this one doesn't exist?:

[](url
[image]
)

Thanks!

MichielVanwelsenaere, 2022-08-03:
Hi!

Looks like you are using an old piece of code there. The whole RS485 functionality got a big refactor some time ago. 

Please check the latest version of the project and adopt where necessary.

drled, 2022-08-04:
Hi, 

Thanks for your quick feedback!
I'm still having the same issue after downloading the newest release and starting from scratch.

After installing the missing libraries,  etc the "FB_RS485_EASTRON_SDM220_1.RequestMaster" error is left.

I've added an archive(Codesys V3.5 SP18) of my testproject
[testproject_20220804.zip](https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3/files/9262114/testproject_20220804.zip)
.

MichielVanwelsenaere, 2022-08-05:
for some reason I can't see any content in the archive you've provided.

Yet, the refactor dropped the method 'RequestMaster' in any RS485 FB so your error makes sense.
Are you sure you also updated your RS485 PRG?
This is more or less how the RUN should look like:
[image]

The init RS485 also requires some updates.
With the refactor, all you need to do is register your RS485 FB against a buscontroller and everything else like acquiring the bus will be handled for you.

drled, 2022-08-06:
Thanks for the feedback, yes I started completely from scratch so RS_485.PRG is also updated.

Seems like you are using the "ecockpit" version and i am using Codesys. In the codesys version the "RequestMaster" is still present.
I will try to adjust the code a bit to get rid of the error.

MichielVanwelsenaere, 2022-08-07:
I had a doublecheck, is this the piece of code you're refering to?
[image]

if so, you can remove it completely. Seems liek I missed to remove this block.

I indeed use écockpit and it's hard to doublecheck everything new in Codesys as I don't have a device running it. 
Yet, every functionality is developped to work on both runtimes.

drled, 2022-08-08:
Yes it's indeed that part of the code :) 
I will try removing that part and let you know the result asap :)

MichielVanwelsenaere, 2022-09-19:
closing due to inactivity.

closed 2022-09-19 by MichielVanwelsenaere