MichielVanwelsenaere/HomeAutomation.CoDeSys3 #124: MQTT switch

meijer3, 2022-10-31:
Hi,

My WRITE_SWITCHES is like:
<pre>
FB_DO_SW_001(OUT=>DO_001 ,PRIO_HIGH:=FALSE,  PRIO_LOW:=FALSE,TOGGLE:=FB_DI_PB_001.SINGLE );
</pre>
I update MQTT with home assistant, to 
`Devices/PLC/House/In/DigitalOutputs/FB_DO_SW_001` : `FALSE` / `TRUE`.
but nothing is changing. I have two questions.
1. I saw in the other issue, that TOGGLE can be set to `FALSE`, but I want both an input pushbutton and overrule it with MQTT. I assumed this was possible, can you confirm that?
2. Do you have any clue why MQTT is not updating the callback function blocks. I am not good in debugging in e!Cockpit (yet... )

MichielVanwelsenaere, 2022-10-31:
Could you provide a copy of your project?
Do you see MQTT hartbeat messages on the broker for the PLC?

meijer3, 2022-11-01:
[HomeAutomation.Device.HomeAutomation.dc8b4b06-db0a-415d-9c9a-bbaa6953bc53.zip](https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3/files/9907326/HomeAutomation.Device.HomeAutomation.dc8b4b06-db0a-415d-9c9a-bbaa6953bc53.zip)
MQTT is working. I can see the changes up to home assistant. 
[image]
[image]

MichielVanwelsenaere, 2022-11-01:
I'm unable to open your project. I'm getting a 'could not open project error' when attempting..

> I want both an input pushbutton and overrule it with MQTT. I assumed this was possible, can you confirm that?

depends on what you call 'overrule', I have pushbuttons configured to output function blocks as well. If someone presses the pushbutton, it will toggle, if a MQTT message is received it will toggle as well. They work side by side.

I'm a bit confused to what is going on with your MQTT topics, looks like a mix of the old MQTT topic structure I factored out some time ago. Are you trying to replicate the update into your own project?

Either way, perhaps the following will clarify somethings for you:

1. publish topic(s): defined in 'PLC_PRG_MAIN' variables
2. hartbeat topic: defined in 'PLC_PRG_MQTT' variables
3. subscribe topic(s) defined in 'PLC_PRG_MQTT' -> 'MQTT_SUBSCRIBE':
[image]
This is also where the topics are coupled to a 'collector' function block which in turn is then used in other function blocks.

Let me now if this helps.

meijer3, 2022-11-02:
I can try it once more, a fresh 'save as'. If nothing works, I could share my screen. (also active on tweakers)
[src.zip](https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3/files/9917404/src.zip)
I started off with your latest `.ecp` and did a 'device replacement' and added 8 DO/DI. Thats all I did.

The topic is like in your screenshot. Here is my screenshot, but online.
[image]

And the online FB_OUTPUT_SWITCH_MQTT>PublishedReceived.
[image]

MichielVanwelsenaere, 2022-11-02:
for certainty I just took a fresh copy as well and everything seems to be working perfectly;
[image]

What I can see in your image is that the 'MQTTSubscribeTopic' starts with 'WAGO-PFC20...', this is part of the old naming and probably the issue here as well. 

Are you sure these are set correctily in your project:
[image]
?

meijer3, 2022-11-02:
I did a `cleanAll` and a new download and it started working again. Indeed I lost the 'WAGO-PFC20...'.
It is working like a charm now. Thank you so much for all the effort you put in this ( the whole repo! )

closed 2022-11-02 by meijer3