MichielVanwelsenaere/HomeAutomation.CoDeSys3 #125: SDM120 support

meijer3, 2022-11-13:
The Eastron SDM220 and SDM630 are supported now. 
Can SDM120 support be added? I am happy to test.

MichielVanwelsenaere, 2022-11-14:
I suppose this request is for 'FB_RS485_EASTRON_SDM_POWER_MQTT'?

MichielVanwelsenaere, 2022-11-14:
I added support for the requested functionality in this branch;
https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3/tree/feature/FB_RS485_EASTRON_SDM_POWER_MQTT-sdm120Support

You can update the function blocks in your project by using this procedure: [link](
https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3/blob/feature/FB_RS485_EASTRON_SDM_POWER_MQTT-sdm120Support/docs/FAQ/Howto_updating_function_blocks.md)

Please let me know when you've tested the functionality.

small fyi; I don't get any notifications for thumbs up on comments, almost missed it.

PR #127 (2022-11-21): FB_RS485_EASTRON_SDM_POWER_MQTT: Adding support for SDM120
## Pull Request Checklist
Check if the tasks below are relevant for your PR, to mark an item as completed use *[X]*.

* [X] Updated the PLCOpen XML export file (only relevant on code change).
* [X] Updated the CoDeSys V3 export file (only relevant on code change).
* [X] Updated or created any relevant documentation.

## Proposed Changes

- Adds support for the Eastron SDM120 on FB_RS485_EASTRON_SDM_POWER_MQTT

## Related issues

- Closes #125
closed 2022-11-21 by MichielVanwelsenaere