MichielVanwelsenaere/HomeAutomation.CoDeSys3 #128: Support for logger

meijer3, 2022-11-20:
I build myself a mqtt logger. Maybe interesting for others? 

An use case is: 
`MqttVariables.MQTT_logger.send('DMX ping Error');`
It outputs to `Devices/PLC/House/debug`

I created a funtion block

<pre>
FUNCTION_BLOCK X_FB_MQTT_LOG 
VAR_INPUT
END_VAR
VAR
	InitMqttDone			:BOOL;
	MqttPublishTopicPrefix	:POINTER TO STRING;
	pMqttPublishQueue		:POINTER TO FB_MqttPublishQueue; 
	MqttQos					:MQTT.QoS;
	MqttRetain				:BOOL;
	MqttMessageReceived		:BOOL;
	
END_VAR	
</pre>

With the same `InitMqtt` methods like other FBs here.
I added a `send` method to the functionblock 

<pre>
METHOD send
VAR_INPUT
	payload			:STRING(128);
END_VAR
----
IF InitMqttDone THEN
pMqttPublishQueue^.AddMessage(
	Payload:= payload,
	Topic := MqttPublishTopicPrefix^,
	Qos := MqttQos,
	MqttRetain := MqttRetain
);
END_IF		
</pre>
And in the `MqttVariables`

<pre>
MQTT_logger							: FB_MQTT_LOG;		
</pre>

meijer3, 2022-11-20:
I like the MQTT debug log because you can transfer it into home-assistant.
The idea is based on the OpenMqttGateway debug topic

MichielVanwelsenaere, 2022-11-21:
how do you transfer it exactly in to Home-Assistant?

Obviously logging is usefull but I wonder if it wouldn't be better to have logging functionality in the individual function blocks (where necessary).

For example; it seems that you build some DMX functionality which performs a ping to an external device. Most likely there is some other functionality concerning DMX as well then just the ping?

In theory, you could build a FB around this, for example FB_DMX_CONTROLLER. This FB could emmit a log via 'FB_DMX_CONTROLLER/log'. The added value here is that if you have two instances of that DMX controller (because you have for example two external dmx drivers) you'll have 'FB_DMX_CONTROLLER_House' & 'FB_DMX_CONTROLLER_Garden' each emmiting their logs via 'FB_DMX_CONTROLLER_House/log' & 'FB_DMX_CONTROLLER_Garden/log' so you know which one is having a ping error exactly? 

I haven't read the OpenMqttGateway docs yet, will try to find some time.

meijer3, 2022-11-25:
You can use a logger in serveral ways. I would only be interested in errors and not complete log files (yet). Therefore I push only errors to mqtt and a simple mqtt sensor would show the last line/error.

<pre>
mqtt:
  sensor:
    - name: "plc_debug"
      object_id: "plc_debug"
      unique_id: "plc_debug"
      state_topic: "Devices/PLC/House/debug"
      qos: 0
</pre>
 I very much like your instance-logger. That would be a great add on.
 Other things. that might be interesting or overkill would be:
- Level (debug/info/error)
- Timestamps
- Line numbers
There are many standards out there, how to build up a log.

meijer3, 2023-01-18:
https://help.codesys.com/webapp/_cds_obj_task_config_sys_event;product=codesys;version=3.5.13.0
Maybe these events are intersting. Like stop and restarts

meijer3, 2023-01-19:
Oh nevermind that long implementation: Simple card
```
   type: logbook
   entities:
     - sensor.plc_log
```

closed 2023-01-26 by MichielVanwelsenaere