MichielVanwelsenaere/HomeAutomation.CoDeSys3 #132: 1/0 instead of TRUE/FALSE

rmeekers, 2023-01-13:
Can the system handle MQTT status messages with `1`/`0` instead of `TRUE`/`FALSE`?
Reason why I'm asking is that the source system that is publishing these messages is only capable of doing `1`/`0`.

MichielVanwelsenaere, 2023-01-13:
Yes that shouldn't be a problem. 
Can it handle TRUE/FALSE messages as subscription messages?

rmeekers, 2023-01-13:
I mean the other way around. I only need one direction. So I publish 1 or 0
and the Wago should switch an output

MichielVanwelsenaere, 2023-01-14:
yes, perfectly possible. As we're talking about switching an output I suppose this refers to 'FB_OUTPUT_SWITCH_MQTT'.

In the method 'PublishReceived' of that FB you can update the 'TRUE' and 'FALSE' values to '1' and '0':
[image]

A few sidenotes though:
- by default the function block will send a TRUE/FALSE message to notify you that the change has been applied.
- the FB wil send a TRUE/FALSE message if the output has been changed by the prio_low/high or toggle inputs. if you don't use them you don't have a problem if you do use them you'll need to figure something out to get these status updates in your system.

Out of curiosity, what system are you using?

rmeekers, 2023-01-19:
Ok thanks Michiel.

I'm using an old Dobiss system. I'm now trying to hook into this systems CAN-bus through an Arduino MKR. It was easier to skip the 0/1 translation to TRUE/FALSE but maybe I should make an extra effort to accomplish that translation.

MichielVanwelsenaere, 2023-01-20:
If you ever want to get rid of the Arduino MKR; you can always reach out to Wago support and ask them for a sample project on now to do it. The docs on their Functionblocks are normally pretty good as well to get you started.

closed 2023-01-20 by MichielVanwelsenaere

rmeekers, 2023-01-21:
> If you ever want to get rid of the Arduino MKR

I've put an Arduino in between because the CAN module from Wago was too expensive (even on the second hand market) – although that was my preferred solution :)