MichielVanwelsenaere/HomeAutomation.CoDeSys3 #133: Idea: Non-linear dimming

meijer3, 2023-01-16:
I see the dimmer FB uses `LIN_TRAFO: Util.LIN_TRAFO`
Most other libraries uses a non-linear like: quadratic cubic or quadruple
https://github.com/Breina/ha-artnet-led#output-correction
[image]

Personally it is hard to time full brightness with a pushbutton right now.
A small delay at the top of the curve would help. I also know some devides (like my headlight) blinks shortly when it is at max.

MichielVanwelsenaere, 2023-01-17:
This would indeed be nice.

MichielVanwelsenaere, 2023-01-24:
can I assign this one to you @meijer3 ?

meijer3, 2023-01-24:
Yes for a first setup, but iam not an expert. :)

MichielVanwelsenaere, 2025-04-30:
Closing due to inactivity.

closed 2025-04-30 by MichielVanwelsenaere