MichielVanwelsenaere/HomeAutomation.CoDeSys3 #165: Unable to Compile Code in 64-Bit Mode (SysFile23 and SysSocket23 Not Compatible)

Kyosuk3, 2024-12-21:
Hello,

First of all, thank you for the great work and effort you’ve put into this repository. It’s a fantastic resource for smart building applications.

I am attempting to use this repository’s code on a WAGO PFC300 PLC. However, when compiling in 64-bit mode with CODESYS v3.5 SP19, I run into a compatibility problem with two system libraries: **SysFile23** and **SysSocket23**. These libraries appear not to support 64-bit compilation.

Would it be possible to update or replace these libraries to allow 64-bit support? Any guidance or workarounds would be greatly appreciated.

Thank you in advance for your help!

Kyosuk3, 2024-12-24:
Thank you for your response,
It seems to be the same version. I also encountered a compilation error due to it being 32-bit.
Given the age of the library, it doesn’t appear to have been designed for compilation with 64-bit instructions.

However, I noticed that there are no direct calls to this library in the program, and it doesn’t seem to have any dependencies with the other libraries either. If I remove it, the program compiles correctly (though I haven’t tested it on the PLC yet).
What is the role of this library in your project?

I had to re-import several libraries and manage some missing dependencies. It’s possible that Wago handles part of the networking differently, which could explain the absence of errors.

Here’s an overview of the libraries used (WagoTypesSocket (1.6.0.0) is missing):
[image]

MichielVanwelsenaere, 2024-12-26:
interesting! honestly, I don't recall. It's possible that it was a dependency for the MQTT library a -long- time ago but that it's no longer required. Let me know if you manage to get things to run on your device.

MichielVanwelsenaere, 2025-04-30:
Closing due to inactivity.

closed 2025-04-30 by MichielVanwelsenaere