MichielVanwelsenaere/HomeAutomation.CoDeSys3 #166: Problem with libraries

elconfa, 2025-04-30:
Hi, I would like to try this project. I don't use a Wago PLC, but Weintek PLCs with Codesis 3.5
Of course, I'll have to delete the Wago libraries and fix the project to get rid of the feature-specific parts with Wago.
The problem is some libraries that I can't install:

PlcServices
StringUtils
FloatinPointUtils

How can I do this?
Thank you for this project and I hope to be able to test it thoroughly before replacing my current PLC for home automation.

Thank you

[image]

elconfa, 2025-04-30:
I was able to load the libraries via another PC.
Can I use this project without Wago PLC? I have the following errors.

Thank you

[image]

[image]

MichielVanwelsenaere, 2025-04-30:
Hmmm, that piece of code is used to extract some device information  like Mac and IP address. It's only used for MQTT discovery purposes. 

I would suggest to replace/remove the troublesome lines so it returns a static value for the Mac and IP address. If you manage to get everything to work you can always revert back to this later on to see if you can get this to work for your specific device. 

The effect of using static non sensible values is quite limited. In Home-Assistant the device will have those non sensible values in the diagnostic section and that's it..

MichielVanwelsenaere, 2025-05-15:
closing due to inactivity.

closed 2025-05-15 by MichielVanwelsenaere